AdaCore/bb-runtimes #1: s-stm32.ads port from STM32F4 to STM32F0

bmilanov, 2017-08-16:
Hi all,

The current src/s-stm32.ads is stm32f4-specific, which is incompatible with stm32f0. Examples of this are:
- Number of TIMCLKs and PCLKs
- Device IDs
- GPIO speed constants
- PLL sources due to clock tree differences

I have copy-pasted-modified the s-stm32.ads to s-stm32f0.ads for this reason.

Should I edit the python build scripts to rename s-stm32f0.ads to s-stm32.ads during RTS/BSP dirtree generation, or are these differences better addressed with one s-stm32.ads file and "if (ARCH == STM32F0)"-like conditions?

lambourg, 2017-08-16:
Hello,

s-stm32 is deemed to be replaced at some point by i-stm32-* packages generated from svd.

In the mean time, the proper way to handle the cpu differences is to use the <base>__<variant>.ad[bs] naming schema, and add the proper handling in arm/cortexm.py. The python library automatically detects the __variant in source files and properly install it as <base>.ad[bs] in the runtime.

Best regards,
- Jerome

bmilanov, 2017-08-16:
Okay, will stick with the naming schema. Thanks!

closed 2017-08-16 by bmilanov