AdaCore/bb-runtimes #2: Runtime for STM32H7x3 ?

FlyingGekkota, 2017-08-25:
Hello,

I wonder if at Adacore you plan to release a runtime (zfp or sfp-ravenscar) targeting the new [STM32H7x3 ](http://www.st.com/en/microcontrollers/stm32h7x3.html?querycriteria=productId=LN2033)?

I am currently trying to port your STM32F7x runtime to the STM32H7x but it is not so obvious (that is really low level stuff :) ) 

Thanks for any information,
Regards 

_PS : Thanks to put your stuff on github, it is really interesting to see / use the various project related to Ada. It seems to me that now, Ada is less and less "obscure" :-)_

lambourg, 2017-08-25:
Hello,

We don't have plans for now on the STM32H7x line.

To adapt the runtime to this new MCU, I suggest you have a look at arm/cortexm.py file and see the work specific to the STM32F769disco support: this should be very similar to what you would need to adapt the runtime to this MCU (same FPU, similar core). Only the clocks might need some adaptation, and of course the memory map specific to the board.

Regards,
- Jerome

PS: thanks :) Always nice to see someone happy with this effort!

closed 2017-08-25 by lambourg

HackInvent, 2018-01-10:
Hi, they are not similar, STM32H7 has 3xPLL, I just did the exercise to add it in the  fork [HackInvent/bb-runtimes](https://github.com/HackInvent/bb-runtimes)
I was obliged to modify the STM32CommonBSP's init because the PLL_Setup existing is not compatible with the STM32H7
Regards,
Badr