AdaCore/bb-runtimes #4: ARMv8 AArch64 ZFP

dloosli, 2017-11-11:
Dear AdaCore Team

I’m quite sure that this is rather a bug in my setup than in your code. But because I’m stuck I opened this issue. 

I’m trying to compile the ZFP runtime for the Raspberry Pi 3 with a custom ARMv8 AArch64 cross compiler (Ubuntu 16.04 host). First, I built a custom ARMv8 AArch64 cross compiler with crosstool-ng. Then I downloaded the community gnat tools and tried to integrate the AArch64 compiler by adding an entry for my cross compiler to the toolchains.py file and a compiler file into the gprconfig directory. After that, the `gprconfig --show-targets` finds the compiler.

To build the BSP / runtime I first of all cloned the bb-runtime repository, the gcc-7.2.0 source code and the community gnat tool into a directory and renamed the gcc-7.2.0 folder to gcc and the gnat folder to gnat (absolutely not sure if that's necessary - but I didn't really understand the ReadMe part with "depends on both gcc and gnat repositories"). Then I generated the BSP for the RPi3 (seems to work). But when I try to build the runtime with `gprbuild --target=mycompiler -P ...`, I always get the following error:

```
libgnat.gpr:1:06: unknown project file: "libgnat_sources"
libgnat.gpr:10:06: unknown package or project "Libgnat_Sources"
libgnat.gpr:21:06: unknown package or project "Libgnat_Sources"
gprbuild: "temp/BSPs/zfp_rpi3.gpr" processing failed
```

I also attached a detailed problem description with my setup to this issue: [adaproblemdescription.pdf](https://github.com/AdaCore/bb-runtimes/files/1463581/adaproblemdescription.pdf)

Thanks in advance for your help!

lambourg, 2017-11-11:
Hello David,

That's expected as the bb-runtimes repository only contains BSPs, but not the full runtime sources. In particular, the ravenscar engine and the bare metal ZFP support is not part of the FSF gcc repository, but is delivered with the GNAT Community edition, under the form of a set of sources controlled by project files (the libgnat_sources project, but also the libgnarl_sources one for ravenscar support).

If you want to play with the Raspberry Pi 2 or 3, you can on the other hand consider the rpi2 runtime that is already part of the latest GNAT Community (e.g. GNAT GPL) targeting ARM (32-bit). The RPi3 is perfectly able to be used in 32-bit mode and is then compatible with the RPi3.

closed 2017-11-11 by lambourg

Fabien-Chouteau, 2017-11-13:
@dloosli 

Hi David,

You can also start from the 32-bit version of the the ZFP for Raspberry Pi 3/2 and compile it for 64-bit. I don't think there will be a lot of things to modify (maybe only arm-eabi/BSPs/cortex-ar/rpi2/zfp/runtime.xml).

Keep us updated :)