AdaCore/bb-runtimes #6: All tasks are blocked indefinitely

osannolik, 2017-12-17:
I have run into a case where all existing tasks gets blocked forever. 

I created an example project in my Ada_Drivers_Library fork that can be run on a STM32F4_Discovery that shows the issue:
https://github.com/osannolik/Ada_Drivers_Library/tree/runtime_issue/examples/STM32F4_DISCO/runtime_issue

In short, the example project has four tasks: two that each are triggered by a separate interrupt, and two that are run periodically. After 1s from application start one of the interrupt triggered tasks is blocked and is never run again. After 46s all remaining tasks gets blocked forever. A more detailed description of the setup and what is happening is described in the project readme. 

Stopping the mcu when all tasks has been blocked shows that it gets stuck in the loop on line 166 in s-bbthqu.adb (procedure Change_Priority), so it seems a task for some reason gets added to the list twice?

lambourg, 2018-01-11:
Hi,

From an initial quick read of your code, you may be missing a Clear_Interrupt at some point. Basically, in the handlers, there's an if <condition> then Clear_Interupt(); end if; that's suspicious.

Clearing the interrupt should be unconditional once you're done with your handler, else the handler will be immediately called back as long as the interrupt condition is met, and I suspect that it's what is happening here.

closed 2018-01-11 by lambourg

osannolik, 2018-01-20:
Well I have tried to clear the interrupts (also clearing all interrupts for the peripheral) unconditionally, see
https://github.com/osannolik/Ada_Drivers_Library/blob/runtime_issue/examples/STM32F4_DISCO/runtime_issue/src/some_tasks.adb

But the issue remains. Any other suggestions?

Fabien-Chouteau, 2018-01-25:
Hi Jonas,

You can try to build the run-time with assertions enabled, this might show where the problem is.
You can also add more checks in the run-time to detect when the faulty situation occurs.

osannolik, 2018-01-28:
Thanks for the reply!

I did a quick test and built the runtime with assertions enabled. Some initial observations:

The first assert that failed was at s-bbprot.adb:112. It seems to check whether the task to be run after a context switch is actually runnable (Threads.Queues.Running_Thread.State = Threads.Runnable). As far as I can tell this occurs as the task Hall_Handler hits the entry-call, and is put to sleep. I need to look closer on why this is so. Call trace:
system.tasking.restricted.stages.task_wrapper
some_tasks.hall_handler
system.tasking.protected_object.single_entry.protected_single_entry_call
system.task_primitives.operations.sleep
system.bb.threads.sleep
system.bb.protection.leave_kernel

Just to see what would happen, I removed that assert and tried again, keeping in mind that any following assertion fails may just be a consequence of the removed assert fail. This time the Post-condition for Change_Priority at s-bbthqu.ads:137 (Thread.Active_Priority = Priority) failed. To get the call trace when this occurs I added checks for that condition just before each return statement so that I could have something to attach my breakpoints to. However, I never managed to get a failure of that condition inside the procedure body, but the post-condition still fails. This is very strange. Either something happens that changes Active_Priority of Thread just inbetween the procedure return and the post-condition checks (this seems extremely unlikely), or the way I debug/understand this is not correct (much more likely).