AdaCore/bb-runtimes #7: The setup_pll is not by default compatible with STM32h7 familly

HackInvent, 2018-01-09:
Hi the STM32H7 family has 3xPLLs, the setup_pll doesn't work without modifying or removing it from the BSPs.
A turn around that I used in a fork is to add in cortexm.py the possibility to select the Stm32CommonBSP's sources from the init of Stm32.

lambourg, 2018-01-11:
Hi, could you point us to the sources of your runtime so that we can have a look and see how we could be support the full family?

Beset regards,
- Jerome

HackInvent, 2018-01-12:
I made this commit to [add SMT32H7](https://github.com/HackInvent/bb-runtimes/commit/168f25ba0dcdb93c857d1d37276c806cd50244e7) in a fork

To make it work I was obliged to remove the setup_pll.ada and modify a little bit the cortexm.py to make it possible.

I don't have the full history why you added Stm32CommonBSP in cortexm.py, perhaps it is not really necessary? we can make all the customization in Stm32._init_ to add new mcu or board

HackInvent, 2018-08-18:
Hi @lambourg 
Any news? if you want I can give a hand to help restructuring?
Badr

lambourg, 2018-09-03:
Sorry for the late reply, but I'll have a look at your fork and see where we could improve portability here. I'll let you know of course.

lambourg, 2018-09-06:
Hi @HackInvent ,

So I finally found time to look at your fork.

On the history behind Stm32CommonBSP, the goal for the bb-runtimes scripts is to generate a source tree that can then be modified/enhanced to easily add new board support without having to re-run from bb-runtimes the whole generation (and thus dive into somewhat complex python infrastructure to do so).

This means that the generated source tree should avoid as much as possible duplicated source files (e.g. avoid having the same source in multiple places).

Now on the stm32h7, there's indeed incompatibilities in the way setup_pll should be implemented.

So your solution of course will work, with the drawback of having setup_pll duplicated on all stm32f* folders. I'll see if things could be more flexible here...

HackInvent, 2019-01-04:
Hi @lambourg did you get a chance to look into a hint about the H7?
We are working on a board based on this chip, if you have any directive we can help to refactor.

HackInvent, 2019-01-22:
Hi @Fabien-Chouteau 
I just made the update as you commented. https://github.com/AdaCore/bb-runtimes/compare/master...HackInvent:master
I kept a temporary folder in the commit in purpose to show the generated BSP.
I also generated/built SMT32F4 example to check if the modification had no regression on the other boards.
Regards,
Badr

HackInvent, 2019-01-22:
@Fabien-Chouteau I add here a commit for Ada_Drivers_Library https://github.com/HackInvent/Ada_Drivers_Library/commit/4230bec358868674fdc471c6e632ffc5bcbff47b to test the configuration with GPIO (simple onboard blink led).
It's working fine.
[image]

HackInvent, 2019-01-30:
I think it was a mistake, I don't remember having an issue with STM32.GPIO.

Fabien-Chouteau, 2019-02-05:
Hi @HackInvent,

Copy paste is generally a bad idea in the long run, it raises the maintenance cost.

closed 2019-08-20 by Fabien-Chouteau

ezwiebel, 2020-06-18:
Hi, 

I know this issue is closed but I'm also looking into developing Ada SW on a nucleo-144 board with STM32H743 MCU and I can't seem to be able to open issues on @HackInvent fork to discuss it. 

Looking at it though, I find it unlikely that you've been able to run your SW on the board correctly as you did not modify any of the low level things like the assembly startup script (start-common.S) for cache initialisation nor all the things pointed out [here](https://github.com/AdaCore/bb-runtimes/blob/community-2018/doc/porting_runtime_for_cortex_m/4_STM32F7.md) for the F7 (same processor Cortex-M7). Also, the flash SVD you use for PLL setup is the one for the F7 and does not point to the right address for the H743... what exactly is done in fact ? Who knows 😄 

Maybe it is enough to pilot a GPIO though... but I wouldn't bet on it working for more complex usage. I'm currently trying to adapt the runtime to the H743 and will need to use DMA/I2C/SPI/USART and everything else. 
Tough work, but I'll let you know how it goes. 
You don't happen to have an "off the shelf" implementation lying around @lambourg @Fabien-Chouteau ? That would be wonderful 😅 

Thanks.

Fabien-Chouteau, 2020-06-19:
Hi @ezwiebel,

> You don't happen to have an "off the shelf" implementation lying around @lambourg @Fabien-Chouteau ? That would be wonderful sweat_smile

All the "off the shelf" BSPs are in this repo already.

ezwiebel, 2020-06-19:
Thanks for the answer. I'm on it, I'll let you know when done if interested.
Cheers

Fabien-Chouteau, 2020-06-19:
Of course, let us know :)

HackInvent, 2020-09-14:
> Looking at it though, I find it unlikely that you've been able to run your SW on the board correctly as you did not modify any of the low level things like the assembly startup script [...]

Not funny, all the low level has been modified to fit with the STM32H7 requirements, you just need to check in the right repository.

ezwiebel, 2020-09-14:
No offense meant...

But still, if you check the flash SVD for H7 in your [repo](https://github.com/HackInvent/bb-runtimes/blob/master/arm/stm32/stm32h7xx/svd/i-stm32-flash.ads) in front of the [H7 reference manual](https://www.st.com/resource/en/reference_manual/dm00314099-stm32h742-stm32h743-753-and-stm32h750-value-line-advanced-arm-based-32-bit-mcus-stmicroelectronics.pdf), the adresses don't match...

16#40023C00# in your SVD, correct value for F7
0x52002000 in datasheet, correct value for H7

Some other things are not consistent with reference manual. As are some startup scripts. I'm not saying it didn't fulfil your needs, just that it is not mature enough to be shipped as a full H7 runtime... And it's not drama, I would've been happy to find something off the shelf cause I'm a lazy bastard, that's it ;)

HackInvent, 2020-09-18:
In my fork I was testing only GPIO/I2C/USART-UART (maybe also SPI)
I didn't update the flash peripheral
In the process I copied/pasted the code from the F7, and updated it using Nucleo board and Adacore IDE debugger

JCGobbi, 2021-11-13:
I don't know if there is still interest on an "off the shelf" runtime and ADL for the STM32H743, but if it is the case, take a look at [Nucleo-STM32H743ZI](https://github.com/JCGobbi/Nucleo-STM32H743ZI). @lambourg @HackInvent