AdaCore/bb-runtimes #8: Trouble using ravenscar_sfp_tms570 on another Dev Board

hama1067, 2018-03-22:
Hello,

I'm trying to port the ravenscar_sfp_tms570 runtime to a XL2-TMS57012 development board.

For the TMS5701224-PGE I changed the Clock_Frequency in s-bbpara.ads to 160MHz. 
I also changed in sys_startup_simple.adb NF (SYS1_PLLCTL1) to 80 so the PLL will generate the right clock frequency.

a-intnam.ads was changed according to Table 6-31 of the datasheet SPNS190B.
In s-bbpara__tms570.ads the interrupt_range was changed to 0 .. 111.

In runtime.xml the Loader Variable was set to Flash. In the linker skript the Memmory Layout was changed to: 
MEMORY
{
  FLASH (rx)  : ORIGIN = 0x00000000, LENGTH = 1310719
  SRAM (rwx)  : ORIGIN = 0x08000000, LENGTH = 192K
}

After this I have no clue, why the board isn't working. 
Any suggestions what I missed?

Best regards,

Martin

P.S.: The link to "Customized run-time" at bb-runtimes/doc/porting_runtime_for_cortex_m/5_CONCLUSION.md is broken.

lambourg, 2018-03-22:
Hello Martin,

Have you tried debugging the startup code to see where things fail?

Best regards,
- Jerome

PS: thanks, will fix.

hama1067, 2018-03-22:
Hello Jerome,

thank you for your quick response!

Not yet. The only thing i can say is, that my main application is not executed, so something in the startup code must go wrong.

Flashing and debugging with openocd is at the moment also an issue:

Flashing image to board.
Could not flash the executable.
[workflow stopped]

I guess I have to build openocd with TI's F021 Flash API by myself first. 
It's not in the prebuild binary anymore.

Well, to flash the microcontroller I'm using TI's UniFlash. 

Best regards,

Martin

lambourg, 2018-03-23:
Good luck then with your investigation! Note that we've refactored this BSP quite a bit in the master branch to allow running on the new TMS570LC43. You may have some inspiration from those changes to get the runtime running on your board.

Best regards,
Jerome

closed 2020-11-25 by Fabien-Chouteau