AdaCore/bb-runtimes #9: Misleading comments in TMS570's board_init.adb src file

martinmarcos, 2018-08-12:
Hello,

I am trying to port bb-runtimes to run on a custom board with a tms570ls1227 mcu. I have been comparing technical reference manuals and HALCoGen generated files from the tms570ls3137 and the tms570ls1227 in order to determine their differences and similarities. So far, start up code is practically the same. I have doubts regarding the cpu errata code, I think TI fixed those problems in the newer chip.

Anyways, I've noticed comments in these [lines](https://github.com/AdaCore/bb-runtimes/blob/237bbad91986aa3e85867dbac82e1a23372b3278/arm/tms570/board_init.adb#L320-L325) of the **board_init.adb** src file for the tms570 are misleading. I have compared them to HALCoGen generated source files, I have checked ARM's Cortex-R4F TRM as well as TI's TMS570LS1227 TRM and I'm pretty certain the code is enabling ECC on TCM(Tightly Coupled Memory) interface B0 and B1 which are connected to RAM memory, not flash as indicated in the comment. Also ACTLR bits being modified are bits 27 and 26 and their names are B1TCMPCEN and B0TCMPCEN respectively, not DICDI and DIB2DI as indicated in the comment.

Flash is connected to TCM interface A, and ECC on this interface is enabled via ACTLR's bit 25, ATCMPCEN.

I found it odd that the ACTLR set instruction in **board_init.adb** is preceded by a DMB instruction and then followed by an ISB instruction while C code generated by HALCoGen didn't include these instructions. Is this inclusion recommended in some documentation perhaps? If so, could you please share it?

Regards.

lambourg, 2018-09-03:
Hello Martin,

Sorry for the late answer, but you know, summertime...
I'll have a look at your various questions, and will get back to you.

Fabien-Chouteau, 2020-11-25:
Hello @martinmarcos, sorry we didn't find time to answer your question.

closed 2020-11-25 by Fabien-Chouteau