AdaCore/bb-runtimes #14: Porting run-time to Arduino Due (SAM3X8E)

BrentSeidel, 2019-01-10:
I am trying to create a runtime so that I can write Ada code that runs on the Arduino Due based on the instructions in the tutorial.  I found the sam4s directory to use as a starting point and copied it to sam3x8e.  I could get that to build.  Once I removed the sam4s svd files and replaced them by sam3x8e generated files, the build failed with the following errors:
--------------------
$ gprbuild -P build/BSPs/ravenscar_sfp_sam3x8e.gpr
Compile
   [Ada]          setup_pll.adb
setup_pll.adb:34:06: "Interfaces.Sam" is not a predefined library unit
setup_pll.adb:35:06: "Interfaces.Sam.Efc" is not a predefined library unit
setup_pll.adb:36:06: "Interfaces.Sam.Pmc" is not a predefined library unit
setup_pll.adb:37:06: "Interfaces.Sam.Sysc" is not a predefined library unit
gprbuild: *** compilation phase failed
-----------------------
Where should I look to find these?

[ravenscar_sfp_sam3x8e.gpr.txt](https://github.com/AdaCore/bb-runtimes/files/2745825/ravenscar_sfp_sam3x8e.gpr.txt)
[ravenscar_sfp_sam4s.gpr.txt](https://github.com/AdaCore/bb-runtimes/files/2745826/ravenscar_sfp_sam4s.gpr.txt)

I've attached the gpr files for both sam4s (which worked) and sam3x8e (which causes the errors).  They are mostly the same except for sam4s->sam3x8e and has floating point set to false.

Any advice would be appreciated.

thanks,
brent

Fabien-Chouteau, 2019-01-10:
Hi @BrentSeidel,

For the run-time you have to call svd2ada with some specific options.

Have a look at this makefile and add the sm3x8e to inside: https://github.com/AdaCore/bb-runtimes/blob/community-2018/arm/sam/svd.mk

Regards,

closed 2019-01-10 by Fabien-Chouteau

BrentSeidel, 2019-01-10:
Hello,

What calls this makefile?  I made the following change to the makefile:
	rm -rf */svd */svdtmp
+	$(SVD2ADA_DIR)/svd2ada $(SVD2ADA_DIR)/CMSIS-SVD/ATMEL/ATSAM3X8E.svd -p Interfaces.SAM -o sam3x8e/svdtmp --boolean
	$(SVD2ADA_DIR)/svd2ada $(SVD2ADA_DIR)/CMSIS-SVD/ATMEL/ATSAM4SD32C.svd -p Interfaces.SAM -o sam4s/svdtmp --boolean
	$(SVD2ADA_DIR)/svd2ada $(SVD2ADA_DIR)/CMSIS-SVD/ATMEL/ATSAMG55J19.svd -p Interfaces.SAM -o samg55/svdtmp —boolean

and reran the gprbuild with the same results.  So I tried to back up a step and reran the build_rts.py with the following results:
$ /usr/bin/python ./build_rts.py --bsps-only --output=build --prefix=arm-eabi/lib/gnat --link sam3x8e
Cannot find source dir for arm/sam/sam3x8e/svd/i-sam-efc.ads

brent