AdaCore/bb-runtimes #17: Cannot generate runtime for STM32F401 using newest svd2ada

ucgosupl, 2019-04-26:
Hello

Recently I was trying to generate runtimes for STM32F401. During this proces I struggled with following issues:

1.  STM32F401 have maximum clock frequency 84MHz, so default files from stm32f40x folder cannot be used.
2. I can't build runtimes generated using newest version of [svd2ada](https://github.com/AdaCore/svd2ada)

First issue was only a little inconsequence as stm32f40x should be general for all MCUs and apparently there are some variants that are not compatible. Copying folder and manual edits were simple in this case.

Second issue was harder. I had following errors:
`s-stm32.adb:53:57: expected an integer type
s-stm32.adb:53:57: found type "Interfaces.Stm32.Rcc.CFGR_SWS_Field"`

`setup_pll.adb:105:36: expected type "Interfaces.Stm32.Rcc.CFGR_SW_Field"
setup_pll.adb:105:36: found type universal integer
setup_pll.adb:219:23: expected type "Interfaces.Stm32.Rcc.PLLCFGR_PLLM_Field"
setup_pll.adb:219:23: found type "Interfaces.Stm32.UInt6"
setup_pll.adb:220:23: expected type "Interfaces.Stm32.Rcc.PLLCFGR_PLLN_Field"
setup_pll.adb:220:23: found type "Interfaces.Stm32.UInt9"
setup_pll.adb:221:23: expected type "Interfaces.Stm32.Rcc.PLLCFGR_PLLP_Field"
setup_pll.adb:221:23: found type "Interfaces.Stm32.UInt2"
setup_pll.adb:222:23: expected type "Interfaces.Stm32.Rcc.PLLCFGR_PLLQ_Field"
setup_pll.adb:222:23: found type "Interfaces.Stm32.UInt4"
setup_pll.adb:272:43: invalid operand types for operator "="
setup_pll.adb:272:43: left operand has type "Interfaces.Stm32.Rcc.CFGR_SWS_Field"
setup_pll.adb:272:43: right operand has type universal integer`

The reason is that format of files generated by svd2ada changed a little and some types are now variadic records depending on As_Array variable. After some fighting I was able to make them work. Here is my fork with this change: https://github.com/ucgosupl/bb-runtimes/commit/57a65c95da2bcf4d9ee33ec2cc0b7390eb0e5a0a

Are you planning to update project to be compatible with new svd2ada? Or maybe you want to support both old and new formats? If you give me some guidelines maybe I will be able to prepare pull request with this change.

Fabien-Chouteau, 2019-04-29:
Hi @ucgosupl ,

The STM32F407 and STM32F401 are probably similar enough that you can use the same code. What is the problem with the max clock frequency?

ucgosupl, 2019-04-30:
Max frequency for STM32F401 is 84MHz. When using default stm32f40x files processor goes to HardFault after setting PLL. Maybe there is a way to set that frequency but it is not documented and I don't know how to do that. Also STM32F401 doesn't have CCM memory so I had to modify linker scripts and asembler startup files.

Fabien-Chouteau, 2019-04-30:
What values did you use in `System.BB.Board_Parameters` (Main_Clock_Frequency, PLLP_Value, etc.)?

ucgosupl, 2019-04-30:
I used:
`   Main_Clock_Frequency : constant := 84_000_000;
   --  Optimal frequency of the system clock. Note that the STM32F411 can go
   --  up to 200 MHz, but all other STM32F40x and STM32F41x MCUs can only do
   --  168 MHz.

   HSE_Clock_Frequency : constant := 8_000_000;
   --  Frequency of High Speed External clock.

   FLASH_Latency : constant := 2;
   PLLP_Value    : constant := 4;
   PLLQ_Value    : constant := 7;`

But had to copy \arm\stm32\stm32f40x and create separate runtime for stm32f401. Is there a way to change clock settings without generating new runtime?

Also please take a look at svd2ada issue described in first post.

Fabien-Chouteau, 2019-05-02:
> FLASH_Latency : constant := 2;
> PLLP_Value : constant := 4;
> PLLQ_Value : constant := 7;`

The numbers look good. Do you still have a HardFault during PLL setup with those numbers?

> But had to copy \arm\stm32\stm32f40x and create separate runtime for stm32f401. Is there a way to change clock settings without generating new runtime?

Unfortunately not at the moment, Ada requires the main clock frequency to be a static value. So the run-time has to be recompiled for each configuration.
 
> Also please take a look at svd2ada issue described in first post.

For that we would have to update our generated files in this repo, to be honest it is not a priority right now.

ucgosupl, 2019-05-02:
I managed to debug with no errors with these settings and generated runtime. Just hoped that it is possible to have more generic solution.

Regarding svd2ada as I said in first post I did this for STM32F401 and I could prepare pull request with similar change for other boards. But I will need some guidelines. If you don't want to do any changes in repo you can just provide information which commit should be used in svd2ada repo to generate svds in old format.

Fabien-Chouteau, 2019-05-20:
> I managed to debug with no errors with these settings and generated runtime. Just hoped that it is possible to have more generic solution.

That great!

> Regarding svd2ada as I said in first post I did this for STM32F401 and I could prepare pull request with similar change for other boards. But I will need some guidelines. If you don't want to do any [...]

Honestly I don't think we will have time to review the changes, so I would not recommend you to spend time on it. We may do this work ourself at some point.

closed 2019-05-20 by Fabien-Chouteau