AdaCore/bb-runtimes #19: Porting run-time to STM32L476RG

BottCode, 2019-10-21:
Hi,

I'm developing an application, written in C and Ada, that exploits [the sensortile supplied in this kit](https://www.st.com/en/evaluation-tools/steval-stlkt01v1.html). This sensortile comes with STM32L476JGY microcontroller, so I need to port the Ada run-time to this target (Am I wrong?).

I'm following [your tutorial](https://github.com/AdaCore/bb-runtimes/tree/community-2019/doc/porting_runtime_for_cortex_m), but I have a problem with the code generated by [svd2ada](https://github.com/AdaCore/svd2ada).

My problem concerns the [PWR peripheral and overdrive section](https://github.com/AdaCore/bb-runtimes/blob/community-2019/doc/porting_runtime_for_cortex_m/3_STM32F469.md#pwr-peripheral-and-overdrive).
(in order to build the new run-time, I've copied the one from stm32f7x).

When I try to build the new run-time, I ran into these errors:
[image]
Here you can see my [s-bbmcpa.ads](https://pastebin.com/4ZeCjSYR) and [s-bbmcpa.adb](https://pastebin.com/TzDMdKB1) files.

These errors are due to the fact that my [svd/i-stm32-pwr.ads](https://pastebin.com/b4MjHNaR) file, generated with svd2ada, does not contain any of these fields.
So, what I need to do, is to edit `s-bbmcpa.ads/b`, but I don't know how.
Why there's no `OD(SW)EN` selector? Perhaps because `SMT32L4` does not allow overdrive? And what about `CSR(1)`? 

I've used [this svd](https://github.com/posborne/cmsis-svd/blob/master/data/STMicro/STM32L4x6.svd) file with svd2ada. I've contacted STM32's support and they said that this SVD file should be correct for the target.

I'm new to embedded development, so I hope I clearly explained my problem.

BottCode, 2019-10-21:
I've set to `null` the `PWR_Overdrive_Enable` procedure in `s-bbmcpa.adb` (supposing that overdrive is not enabled in stm32l476rg) and removed `Is_PWR_Stabilized` function in `s-bbmcpa.ads` (I'm not sure if this removal could lead to some problem), but now compiler said `cannot generate code for file s-stm32.ads (package spec)`, which is a file common to all STM32 run-time.

Fabien-Chouteau, 2019-10-21:
> I'm developing an application, written in C and Ada, that exploits [the sensortile supplied in this kit](https://www.st.com/en/evaluation-tools/steval-stlkt01v1.html). This sensortile comes with STM32L476JGY microcontroller, so I need to port the Ada run-time to this target (Am I wrong?).

You are not wrong, but which run-time do you need depends on what you want to achieve ([more info here](https://github.com/AdaCore/Ada_Drivers_Library/blob/master/docs/porting_to_a_new_platform.md)). 

If you have an application mostly in C, where you just want to implement some parts in Ada then you probably don't need a Ravenscar run-time. If you don't need to have different tasks, you probably don't need a Ravenscar run-time.

> I'm following [your tutorial](https://github.com/AdaCore/bb-runtimes/tree/community-2019/doc/porting_runtime_for_cortex_m), but I have a problem with the code generated by [...]

Different STM32 microcontrollers have different clock controls. You have to adapt the code to the SMT32L4.

Fabien-Chouteau, 2019-10-21:
@BottCode if you tell me more about your application I can tell you if you need to do the port of Ravenscar or not.

BottCode, 2019-10-21:
Hi @Fabien-Chouteau 

> If you have an application mostly in C, where you just want to implement some parts in Ada then you probably don't need a Ravenscar run-time. If you don't need to have different tasks, you probably don't need a Ravenscar run-time.

This is exactly my scenario. Moreover, my application does not need task support. In fact I'm working on porting `ZFP` run-time.

> Different STM32 microcontrollers have different clock controls. You have to adapt the code to the SMT32L4.

Do you mean to [set the correct clock frequencies](https://github.com/AdaCore/bb-runtimes/blob/community-2018/doc/porting_runtime_for_cortex_m/3_STM32F469.md#system-clocks) or something else? I've already set `Main_Clock_Frequency` to `80_000_000` and `HSE_Clock_Frequency` to `8_000_000`.

Fabien-Chouteau, 2019-10-22:
Hi @BottCode ,

> This is exactly my scenario. Moreover, my application does not need task support. In fact I'm working on porting `ZFP` run-time.
> 

In that case we have a much simpler solution for you. We did not advertised it yet because there is still things to iron out.

In GNAT Community 2019 there is a ZFP run-time for each kind of ARM Cortex-M core (M0, M0+, M1, M3, M4, M4F, M7F, M7DF). These run-times do no have a linker script or startup code (crt0.S), but this shouldn't be a problem in your case because you integrate Ada in a C application. For the STM32L476 the run-time to use is `zfp-cortex-m4f`.

So what I recommend is to create a project file like this one:
```
project Ada_Module is
  
  for Library_Kind use "static";
  for Library_Name use "adamodule";
  
  for Target use "arm-elf";
  for Runtime ("Ada") use "zfp-cortex-m4f";

  for Source_Dirs use ("src");
  for Object_Dir use "obj";
  for Library_Dir use "lib";
  for Create_Missing_Dirs use "True";
end Ada_Module;
```

Put your Ada package with function exported to C ([tutorial here](https://learn.adacore.com/courses/intro-to-ada/chapters/interfacing_with_c.html)) in the `src` directory and compile it with:
```
$ gprbuild -P ada_module.gpr
```

You will get a `lib/libadamodule.a` static library that you can link with your C code.

> Do you mean to [set the correct clock frequencies](https://github.com/AdaCore/bb-runtimes/blob/community-2018/doc/porting_runtime_for_cortex_m/3_STM32F469.md#system-clocks) or something else? I've [...]

Yes but not only. The clock is produced by a PLL that needs to be configured, there is also the timing of the flash memory.

BottCode, 2019-10-29:
> In that case we have a much simpler solution for you. We did not advertised it yet because there is still things to iron out.
>
> In GNAT Community 2019 there is a ZFP run-time for each kind of ARM [...]

This fits perfectly with what I have to do and it is also a simpler solution. Thanks a lot!

closed 2019-10-29 by BottCode

BottCode, 2019-10-29:
> > In that case we have a much simpler solution for you. We did not advertised it yet because there is still things to iron out.

Can you tell me something about this thing to iron out?

> This fits perfectly with what I have to do and it is also a simpler solution. Thanks a lot!

Fabien-Chouteau, 2019-10-30:
> Can you tell me something about this thing to iron out?

It is a tool that generates linked script and startup code (crt0) based on the characteristics of the hardware (memory size for instance).

> This fits perfectly with what I have to do and it is also a simpler solution. Thanks a lot!

Awesome :)

BottCode, 2019-11-04:
Hi @Fabien-Chouteau,

I'd like to ask another question related to `zfp-cortex-m4f`.

> In GNAT Community 2019 there is a ZFP run-time for each kind of ARM Cortex-M core (M0, M0+, M1, M3, M4, M4F, M7F, M7DF). These run-times do no have a linker script or startup code (crt0.S), but this shouldn't be a problem in your case because you integrate Ada in a C application.

`zfp-cortex-m4f` runtime also includes `System.Memory` package (`s-memory.adsb` file). In this package we can find the following definition:

    package body System.Memory is
        use System.Storage_Elements;

        Heap_Start : Character;
        for Heap_Start'Alignment use Standard'Maximum_Alignment;
        pragma Import (C, Heap_Start, "__heap_start");
        --  The address of the variable is the start of the heap

        Heap_End : Character;
        pragma Import (C, Heap_End, "__heap_end");
        --  The address of the variable is the end of the heap

        .
        .
        .

    end System.Memory;
 
What is this package good for? 
As you said, this runtime does not provide any linker script or startup code and `__heap_start` and `__heap_end` should be defined within the linkerscript, isn't it?

These are the steps I've taken (remind my app is mainly written in C):

1. Create a new runtime based on `zfp-cortex-m4f` in order to [include the Math Library](http://docs.adacore.com/live/wave/gnat_ugx/html/gnat_ugx/gnat_ugx/customized_run-time_libraries.html#tailoring-the-run-time-project). I've called it `zfp-cortex-m4f-math`.
2. Built it (succesfully).
3. Compile my Ada module (as a static library) on `zfp-cortex-m4f-math` runtime (succesfully).
4. Link it to my C application with **no** success.

The reason which lead step 4 to fail is `undefined reference to __heap_start` (and `__heap_end`). 
My application actually does not use dynamic memory (so no use of `malloc`, `calloc` or `free`) and each variable/object size is statically known. It follows that my application does not use Heap.

So, `__heap_start` and `__heap_end` should be useless for my scenario. What I've done is rebuilt `zfp-cortex-m4f-math` runtime without `System.Memory` package. At this point, I can link my Ada library to C application succesfully and everything seems works!

So, are  `__heap_start` and `__heap_end` useful in order to compile a library with no use of the Heap? How can they be defined **without** a linker script?

Thanks for your kindness shown so far.

jklmnn, 2019-11-05:
`__heap_start` and `__heap_end` are imported symbols so your application needs to provide them to the runtime. With these you can specify a specific memory region that your application can use as heap. This memory region can theoretically reside anywhere but I think a common approach is to declare a chunk of BSS and then use the chunks start and end addresses to define the heap with these symbols.

> How can they be defined without a linker script?

This is how you would do a heap in BSS memory (correct me if I forgot something).
```Ada
package Heap is

   type Byte is mod 2 ** 8;
   type Buffer is array (Integer range <>) of Byte;

   Heap : Buffer (1 .. 1024 * 1024);

   Heap_Start : System.Address := Buffer (Buffer'First)'Address with
      Export,
      External_Name => "__heap_start";

   Heap_End : System.Address := Buffer (Buffer'Last)'Address with
      Export,
      External_Name => "__heap_end";

end Heap;
```

Since you don't need heap memory at all removing `System.Memory` seems to be the safest way to fix this (as it would fail at compile time if anything uses heap implicitly).

Fabien-Chouteau, 2019-11-05:
> What is this package good for?

It provide `malloc` (actually `new` in Ada) dynamic memory allocation. Note that in the ZFP profile it is not possible to free the allocated memory.

> As you said, this runtime does not provide any linker script or startup code and `__heap_start` and `__heap_end` should be defined within the linkerscript, isn't it?

Yes they should come from the linker script. Sometimes they are defined with slightly different names like `_heap_start`, or `heap_start`. Can you change your linker script?

>     1. Create a new runtime based on `zfp-cortex-m4f` in order to [include the Math Library](http://docs.adacore.com/live/wave/gnat_ugx/html/gnat_ugx/gnat_ugx/customized_run-time_libraries.html#tailoring-the-run-time-project). I've called it `zfp-cortex-m4f-math`.

What I have done once is to copy the "Math Library" packages in the source of my project, but without the `Ada` root package. So `Ada.Numerics.Elementary_Functions` becomes `Numerics.Elementary_Functions`. This allowed me to have them in the application code rather than the run-time.

> So, are `__heap_start` and `__heap_end` useful in order to compile a library with no use of the Heap? How can they be defined **without** a linker script?

See my comment on @jklmnn code.

Regards,

Fabien-Chouteau, 2019-11-05:
> This is how you would do a heap in BSS memory (correct me if I forgot something):

Thank you @jklmnn , I think this code will almost work but not quite.

The reason is that the important thing here is the address of the `__heap_*` symbols not their value. So we would have to do something like this:
```ada
   Heap_Start : Byte
     with 
       Address => Heap (Heap'First)'Address,
       Export,
       External_Name => "__heap_start";
```
But it is not allowed to export [an object with an address clause](https://gcc.gnu.org/onlinedocs/gnat_rm/Address-Clauses.html). So I am not sure it is even possible to export the `__heap_*` symbols from Ada sources.

BottCode, 2019-11-05:
> What I have done once is to copy the "Math Library" packages in the source of my project, but without the Ada root package. So Ada.Numerics.Elementary_Functions becomes Numerics.Elementary_Functions. This allowed me to have them in the application code rather than the run-time.

It sounds good.

Thanks @Fabien-Chouteau and @jklmnn for your answer!

jklmnn, 2019-11-05:
> But it is not allowed to export an object with an address clause. So I am not sure it is even possible to export the __heap_* symbols from Ada sources.

Well one could, in theory, create a package with one symbol before and one symbol after the actual heap end export these symbols. Although this highly depends on the compiler not changing the memory layout of the package.

OneWingedShark, 2019-11-05:
You could try this:

    Heap : Buffer (1 .. 1024 * 1024);

    Heap_Start : constant System.Address := Heap (Heap'First)'Address with
      Export,
      External_Name => "__heap_start";

    Heap_End : constant System.Address := Heap (Heap'Last)'Address with
      Export,
      External_Name => "__heap_end";

Alternatively, make use of renames:

    Heap_First : Byte renames Heap(Heap'First);
    Heap_Last  : Byte renames Heap(Heap'Last);

    Heap_Start : constant System.Address := Heap_First'Address with
      Export,
      External_Name => "__heap_start";

    Heap_End : constant System.Address := Heap_Last'Address with
      Export,
      External_Name => "__heap_end";

Fabien-Chouteau, 2019-11-06:
@OneWingedShark you will have the same results as above. The value of the symbols doesn't matter, it is the address that counts.

commit 4522f961 (2023-11-08): Merge branch 'discover-gnat-llvm-source-descriptor' into 'master'

Discover the source descriptor in GNAT-LLVM installations

Closes #19

See merge request eng/toolchain/bb-runtimes!53