AdaCore/bb-runtimes #20: Can I generate a generic runtime ?

NicoPy, 2020-01-26:
Hi,

I'm in the process of learning ADA to, eventually, replace C in my embedded projects.
The language itself is important, but the full ecosystem is import too.

I understand the usefulness of runtimes in ADA toolchain. However, the way they are managed does not suit my needs. I don't want a toolchain locked on a board with a specific configuration.

I use the same microcontroller on many different boards. Each board makes use of the microcontroller in a different way (pinning, clocking...). Further more, each board uses 2 binaries : One for a bootloader and one for the application.
With current bb-runtimes model, I have to create 2 runtimes for each board. Furthermore, the clock configuration often evolve during the development, meaning modifying the 2 associated runtimes.
This is not manageable.

Is there a way to generate generic runtimes ?
For example, for a cortex M4f core, generate "zfp-m4f", "ravenscar-sfp-m4f" and "ravenscar-full-m4f".
This way, with 3 runtimes, I can code everything I want easily. 

Regards,
Nicolas

Fabien-Chouteau, 2020-01-27:
Hello @NicoPy ,

There are different reasons for this lack of generic Ravenscar run-times. Some of the board configuration parameters have to be known at compile time (CPU clock, external crystal frequency, etc). So each run-time must be compiled for a board specifically. We can do generic ZFP run-time but no generic Ravenscar run-times (at least until we find a better solution).

If don't absolutely need the Ravenscar features I would recommend the generic ZFP and also have a look at this tool: [startup-gen](https://github.com/AdaCore/startup-gen).

NicoPy, 2020-01-27:
@Fabien-Chouteau 
Bonjour Fabien,

> Some of the board configuration parameters have to be known at compile time (CPU clock, external crystal frequency, etc).

That's what I've understood. 
This requirement is for delays, timeouts and task scheduling. Right ?
I didn't had a look at source code for now, but isn't it possible to workaround this requirement with a configuration function setting a callback and the time between each call to the callback ?

Regards,
Nicolas

Fabien-Chouteau, 2020-01-27:
The Ada language requires that `Ada.Real_Time.Time_Unit` is known at compile time, so using callbacks will not work here.

We are aware of this constraint and we agree that it is annoying, so far we are limited by the requirements of the language.

NicoPy, 2020-01-27:
Could `Ada.Real_Time.Time_Unit` be an external object from the runtime point of view ?
Then the application would instantiate it and the linker would place it in the right location.

Fabien-Chouteau, 2020-01-27:
`Ada.Real_Time.Time_Unit` has to be a compile time value (named number).

NicoPy, 2020-01-28:
> Ada.Real_Time.Time_Unit has to be a compile time value (named number).

Ok.
What about compiling the Ravenscar sources with the application ?

Fabien-Chouteau, 2020-01-29:
> What about compiling the Ravenscar sources with the application ?

Some of the run-time files can be compiled with the application, but not all of them so that won't work either. Again we are aware of this constraint and if there was an easy option we would already have implemented it.

NicoPy, 2020-01-30:
> Again we are aware of this constraint and if there was an easy option we would already have implemented it.

I have no doubt on this. This does not mean others can't find a solution.
Can you provide me with a document explaining the 'whole thing' ? For example, what each file/folder is used for ?
I'd like to understand how it works.

Fabien-Chouteau, 2020-02-11:
It means there is no simple solution that fits our requirements.

All the documentation is either in this repo or provided with GNAT, e.g. https://docs.adacore.com/gnathie_ug-docs/html/gnathie_ug/gnathie_ug.html

NicoPy, 2020-02-17:
Thanks for the link Fabien. This is very interesting.

corecode, 2020-05-26:
Why does `Ada.Real_Time.Time_Unit` depend on the target (or target configuration)?  RM D.8 doesn't seem to imply that `Time_Unit` has to be equal to the clock frequency.

Fabien-Chouteau, 2020-05-28:
```
Time_Unit is the smallest amount of real time representable by the Time type;
it is expressed in seconds.
```

In our implementation this the time between two clock ticks, so it is directly related to the clock frequency.

corecode, 2020-05-28:
Ah so this is an artifact of the implementation, not a requirement of the standard.

Fabien-Chouteau, 2020-05-28:
In practice, having anything else than a Time_Unit directly coming fromthe clock frequency will require a lot more code and overhead on bare-metal.

NicoPy, 2020-08-18:
I finaly found a solution to my problem.
I modified build scripts to be able to :
- easily generate a new runtime with the correct core frequency.
- easily change the core frequency of a runtime

1) The need for several runtimes with different core frequencies implies creating a new runtime for each frequency.
Creating a new runtime means adding a lot of files and directories in the runtime sources.
This is not a manageable.
I added a switch to argparse to be able to add a variant name to the generated runtime. 
For example, I have a `mk22f51212` runtime. By passing `-n test` to build.py, the generated runtime is `ravenscar-sfp-mk22f51212_test` for a ravenscar sfp profile.
This way, I can generate several runtimes with the same sources very easily.

2) I need to be able to set/change the core frequency of the runtime easily.
For this, I added another switch to argparse and added a function to modify the frequency in the installed sources. By passing `-s 8000000` to build.py, the generated runtime is modified with the specified frequency.
For this to work, the core frequency is defined like this :
`   Clock_Frequency : constant := 8_000_000;     --  %sys_clk_freq%`
Please note the `%sys_clk_freq%`tag which is used to detect the file line to be modified.

3) Each runtime is quite resource (disk) hungry. 
I added a switch to generate only the zfp, ravenscar-sfp, ravenscar-full or all runtime types.

4) The application and the runtime must use the same core frequency.
I add an `assert` in each application to verify the runtime frequency is correctly set.

The help message of my nex build.py is :
```
usage: build_rts.py [-h] [-v] [-f] [--rts-src-descriptor RTS_SRC_DESCRIPTOR] [--gen-doc] [-o OUTPUT] [-l] [-b] [--build-flags BUILD_FLAGS]
                    [-t {zfp,ravenscar-sfp,ravenscar-full,all}] [-n VARIANT_NAME] [-s SYS_FREQ]                                           
                    target [target ...]                                                                                                   
                                                                                                                                          
positional arguments:                                                                                                                     
  target                List of target boards to generate runtimes for                                                                    
                                                                                                                                          
optional arguments:                                                                                                                       
  -h, --help            show this help message and exit                                                                                   
  -v, --verbose         Verbose output                                                                                                    
  -f, --force           Forces the installation by overwriting any pre-existing runtime.                                                  
  --rts-src-descriptor RTS_SRC_DESCRIPTOR                                                                                                 
                        The runtime source descriptor file (rts-sources.json)                                                             
  --gen-doc             Generate the documentation                                                                                        
  -o OUTPUT, --output OUTPUT                                                                                                              
                        Where built runtimes will be installed                                                                            
  -l, --link            Use symlinks instead of copies when installing                                                                    
  -b, --build           Build the runtimes                                                                                                
  --build-flags BUILD_FLAGS                                                                                                               
                        Flags passed to gprbuild                                                                                          
  -t {zfp,ravenscar-sfp,ravenscar-full,all}, --rts-type {zfp,ravenscar-sfp,ravenscar-full,all}                                            
                        RTS build type                                                                                                    
  -n VARIANT_NAME, --variant-name VARIANT_NAME                                                                                            
                        Set rts variant name                                                                                              
  -s SYS_FREQ, --sys-freq SYS_FREQ                                                                                                        
                        Set system core frequency                                                                                         
```

Here is an example usage (windows command line) :
`build_rts.py -o c:\gnat\2020-arm-elf\arm-eabi\lib\gnat -b mk22f51212 -t ravenscar-sfp -n bootloader -s 48000000`

This will generate a ravenscar sfp runtime with a core frequency of 48MHz. This runtime is generated in a directory named ravenscar-sfp-mk22f51212_bootloader.

Changing the core frequency of this runtime is as simple as re-running the command with a different -s option value.
It is also very easy to generate other runtime variants by changing the -n option value.

If you are interested in my solution, I can create a pull-request. Or just publish my code here (only 2 files modified).

Nicolas

Fabien-Chouteau, 2020-08-28:
Hello @NicoPy ,

This is very interesting. I think that if we integrate this kind of source modification system it would have to be more "generic".
To be able not only to change one constant, but as many as we want to define (stacks sizes, number of interrupts, etc.).

closed 2020-08-28 by Fabien-Chouteau

NicoPy, 2020-08-28:
Sure. It can be extended to other parameters.

Note concerning the parameter tag :

 1 Notation with a default value :
    `Clock_Frequency : constant := 8_000_000; -- %sys_clk_freq%`

2 Notation without default value (force the use of build option else compile gives an error) :
   `Clock_Frequency : constant := %sys_clk_freq%;`