AdaCore/bb-runtimes #24: community 2020 : runtime build problems

NicoPy, 2020-06-02:
Hi,

I'm experimenting with community 2020, trying to build a runtime.
With the command `c:\python38\python.exe build_rts.py --output=build/st --bsps-only stm32f429disco` I get the following error :
```
usage: build_rts.py [-h] [-v] [-f] [--rts-src-descriptor RTS_SRC_DESCRIPTOR] [--gen-doc] [-o OUTPUT] [-l] [-b] [--build-flags BUILD_FLAGS]
                    target [target ...]
build_rts.py: error: unrecognized arguments: --bsps-only
``` 

Removing `--bsp-only` option, makes the command work :
```
install runtime sources for stm32f429disco
runtimes successfully installed in build\st
```

Next, with the build command : `gprbuild -P build\st\zfp-stm32f429disco\runtime_build.gpr -j0 -f`, I get the following error :
```
runtime_build.gpr:12:23: library directory "Z:\ada\runtimes_src\bb-runtimes\build\st\zfp-stm32f429disco\adalib\" does not exist
runtime_build.gpr:13:22: object directory "obj" not found
gprbuild: "build\st\zfp-stm32f429disco\runtime_build.gpr" processing failed
```
After creating the `obj` and `adalib`directories manualy, re-running the command succeeds.

README.md should be updated.

Fabien-Chouteau, 2020-06-04:
Thank you for the report @NicoPy , we will update the readme.

NicoPy, 2020-06-04:
For missing directory, `-p` option should be specified instead of creating directories manually.

Next, I tried to use a runtime I've made . Once installed, I can't select it in GPS. GPS does not show it in the `Toolchain` configuration box (it was working with with Community Editon 2019).

To install my runtime, I do `gprinstall -f -p -P build\kinetis\zfp-mk22f51212\runtime_build.gpr --prefix=c:\gnat\2020-arm-elf\arm-eabi\zfp-mk22f51212`.

This creates a _zfp-mk22f51212_ directory in `c:\gnat\2020-arm-elf\arm-eabi` with the same structure as in other runtimes (in Community Edition 2019 the structure was different).

I have not found any solution for now.

Fabien-Chouteau, 2020-06-04:
Hello @NicoPy ,

The README is now up to date.

> Next, I tried to use a runtime I've made . Once installed, I can't select it in GPS. GPS does not show it in the `Toolchain` configuration box (it was working with with Community Editon 2019).

This is because the new build script doesn't install in the compiler by default, and doesn't require `gprinstall`.

You should do:
```
./build_rts.py --output=c:\gnat\2020-arm-elf\arm-eabi\lib\gnat --build
```

> `c:\gnat\2020-arm-elf\arm-eabi`

This is not the correct folder for run-time, it should be `c:\gnat\2020-arm-elf\arm-eabi\lib\gnat`.

Regards,

closed 2020-06-04 by Fabien-Chouteau

NicoPy, 2020-06-04:
> This is because the new build script doesn't install in the compiler by default, and doesn't require gprinstall.

Ok. 

> This is not the correct folder for run-time, it should be c:\gnat\2020-arm-elf\arm-eabi\lib\gnat.

I've discovered this late this afternoon.

Thanks for your support.