AdaCore/bb-runtimes #26: Community 2020 : build_rts.py bug

NicoPy, 2020-06-05:
There is a bug in `build_rts.py` at line 273.
This line content is `os.path.relpath(dest, os.getcwd()))`.
This code works only when the `bb-runtimes` sources are located on the same drive letter than the GNAT arm compiler.
In my case, I work in the _z_ drive while the compiler is installed in the _c_ drive.
I get the following error :
``` python
Traceback (most recent call last):
  File "build_rts.py", line 277, in <module>
    main()
  File "build_rts.py", line 273, in main
    os.path.relpath(dest, os.getcwd()))
  File "c:\python38\lib\ntpath.py", line 703, in relpath
    raise ValueError("path is on mount %r, start on mount %r" % (
ValueError: path is on mount 'c:', start on mount 'Z:'
``` 
Please note that this error happens at the very end of the script while printing a successful message. So this has no impact on the build process.

One simple solution is to replace 
```
    print("runtimes successfully installed in %s" %
          os.path.relpath(dest, os.getcwd()))
```
with 
```
    print("runtimes successfully installed in %s" % dest)
```

Fabien-Chouteau, 2020-06-08:
Hi @NicoPy,

Thank you for the report, we will probably use your solution.

Fabien-Chouteau, 2020-06-08:
This is now fixed.

Thank you @NicoPy

closed 2020-06-08 by Fabien-Chouteau

NicoPy, 2020-06-08:
Thanks @Fabien-Chouteau.