AdaCore/bb-runtimes #30: How did you test zynq7000 runtime?

BottCode, 2020-08-04:
Hi,

I'm editing the `ravenscar_full_zynq7000` run-time for my project and I'm wondering how did you test it.
In GNAT CE 2018 there's no an emulator for this board  and even the one provided by Xilinx (i.e. a QEMU fork) does not support the dual-core version of this board. Just the single-core version. Unfortunately, even the original QEMU has the same issue.

Did you use an emulator?

Thanks for you support!

Fabien-Chouteau, 2020-08-05:
Hi @BottCode,

We have a fork of QEMU with some custom adaptations: https://github.com/AdaCore/qemu/tree/qemu-stable-5.0.0

Regards,

closed 2020-08-05 by Fabien-Chouteau

BottCode, 2020-08-05:
Does your qemu version support the zynq7000 dual-core version?
Command `qemu-system-arm -M xilinx-zynq-a9 -smp 2` returns:
 `qemu-system-arm: Invalid SMP CPUs 2. The max CPUs supported by machine 'xilinx-zynq-a9' is 1`

Fabien-Chouteau, 2020-08-05:
I looked at the run-time, it doesn't support running on two core.

For QEMU we use `qemu-system-arm -nographic -M xilinx-zynq-a9 -semihosting -no-reboot -kernel <PROGRAM>`.

BottCode, 2020-08-05:
So, the only way to test my dual version run-time is to execute it on the target and watch what happen with the debugger, isn't it? Is there some additional (and more handy) choice?

Fabien-Chouteau, 2020-08-05:
You can add support for SMP in QEMU or look for another emulator that supports it.