AdaCore/bb-runtimes #42: Update build instructions in README.md.

BrentSeidel, 2020-10-15:
--
Using the following command as indicated in the README.md file

./build_rts.py --output=temp --build sam4s

 ends in failure with the following traceback:

install runtime sources for sam4s
Traceback (most recent call last):
  File "./build_rts.py", line 274, in <module>
    main()
  File "./build_rts.py", line 251, in main
    dest, rts_descriptor=args.rts_src_descriptor)
  File "/Volumes/External - Data/Users/brent/Development/Ada/bb-temp/bb-runtimes/support/bsp_sources/installer.py", line 159, in install
    runtime_sources = self._find_rts_sources(destination, rts_descriptor)
  File "/Volumes/External - Data/Users/brent/Development/Ada/bb-temp/bb-runtimes/support/bsp_sources/installer.py", line 136, in _find_rts_sources
    assert ret is not None, "Cannot find %s" % rts_json_file
AssertionError: Cannot find rts-sources.json
--
There is a command get_rts_sources.py which looks like it might create the rts-sources.json file, but when I run it as:

./gen_rts_sources.py --output=temp --rts-profile ravenscar-sfp

it also ends with the following traceback:

Traceback (most recent call last):
  File "./gen_rts_sources.py", line 93, in <module>
    main()
  File "./gen_rts_sources.py", line 88, in main
    rts_sources=sources, rts_scenarios=all_scenarios)
  File "/Volumes/External - Data/Users/brent/Development/Ada/bb-temp/bb-runtimes/support/rts_sources/__init__.py", line 212, in __init__
    self.add_sources(key, srcs)
  File "/Volumes/External - Data/Users/brent/Development/Ada/bb-temp/bb-runtimes/support/files_holder.py", line 162, in add_sources
    self.add_source(dir, src)
  File "/Volumes/External - Data/Users/brent/Development/Ada/bb-temp/bb-runtimes/support/files_holder.py", line 152, in add_source
    self.add_source_alias(dir, base, src)
  File "/Volumes/External - Data/Users/brent/Development/Ada/bb-temp/bb-runtimes/support/files_holder.py", line 134, in add_source_alias
    self.dirs[dir].append(FilePair(dst, src))
  File "/Volumes/External - Data/Users/brent/Development/Ada/bb-temp/bb-runtimes/support/files_holder.py", line 59, in __init__
    "Error: source file %s not found in gnat" % src
AssertionError: Error: source file libgnat/a-assert.ads not found in gnat
--
The README.md file should be updated to list the current build procedure and what assumptions are being made by the tools.

Fabien-Chouteau, 2020-10-16:
Hi @BrentSeidel ,

`rts-sources.json` should be found in your GNAT Community install: `<INSTALL>/arm-eabi/lib/gnat/rts-sources.json`.
Is the compiler in `PATH`?

BrentSeidel, 2020-10-16:
I did find its-sources.json in the GNAT ARM Community install.  Both the ARM and the native compilers are in $PATH.  I've tried ./build_rts.py on both MacOS and Windows 10 and get similar results about rts-sources.json.  Do I need to copy it somewhere else?

Fabien-Chouteau, 2020-10-20:
The script uses `gprls` to find it.

What is the output of this command in your context?
```
gprls -v --target=arm-eabi
```

BrentSeidel, 2020-10-20:
Hi @Fabien-Chouteau,

The output from gprls -v --target=arm-eabi is:
gprconfig: language 'ada', target 'arm-eabi', default runtime

GPRLS Community 2020 (20200818) (x86_64-apple-darwin17.7.0)
Copyright (C) 2015-2020, AdaCore

Source Search Path:
   /Volumes/External - Data/Users/brent/Development/Ada/bb-runtimes//

Object Search Path:
   /Volumes/External - Data/Users/brent/Development/Ada/bb-runtimes//

Project Search Path:
   <Current_Directory>
   /usr/share/gtkada/lib/gnat
   /Volumes/External - Data/Users/brent/opt/GNAT/2020/arm-eabi/lib/gnat
   /Volumes/External - Data/Users/brent/opt/GNAT/2020/arm-eabi/share/gpr
   /Volumes/External - Data/Users/brent/opt/GNAT/2020/share/gpr
   /Volumes/External - Data/Users/brent/opt/GNAT/2020/lib/gnat

For reference, my $PATH is (with non-gnat items obscured:
@@@:/Users/brent/opt/GNAT/2020/bin:/Users/brent/opt/GNAT/2019/bin:/Users/brent/opt/GNAT/2020-arm-elf/bin:/Users/brent/opt/gnat/2019-arm-elf/bin:/usr/local/bin:@@@:@@@:@@@:@@@:@@@:@@@

Note that there is a symlink from /Users/brent/opt to /Volumes/External - Data/Users/brent/opt.

Is the ARM-ELF installation expected to be installed on top of the GNAT 2020 installation or in a separate directory?  I have it in a separate directory.

Fabien-Chouteau, 2020-10-21:
The script `support/bsp_sources/installer.py` is looking for `rts-sources.json` all the `Project Search Path:` directories:
```
/usr/share/gtkada/lib/gnat
/Volumes/External - Data/Users/brent/opt/GNAT/2020/arm-eabi/lib/gnat
/Volumes/External - Data/Users/brent/opt/GNAT/2020/arm-eabi/share/gpr
/Volumes/External - Data/Users/brent/opt/GNAT/2020/share/gpr
/Volumes/External - Data/Users/brent/opt/GNAT/2020/lib/gnat
```

Fabien-Chouteau, 2020-10-21:
You can add some prints in `support/bsp_sources/installer.py` to have more info. For instance:
```python
    def _find_rts_sources(self, destination, descriptor):
        """Find the runtime sources and the json file that describes them.
        """
        # First look in the relative path
        rts_json_file = 'rts-sources.json'

        ret = None
        if descriptor is not None:
            # First: take the one given on the command line if any
            ret = os.path.abspath(descriptor)
        else:
            # Next: try a relative path from the install directotry
            fname = os.path.join(destination, rts_json_file)
            if os.path.exists(fname):
                ret = os.path.normpath(fname)

        print(str(ret))
        if ret is None:
            # Finally: Use gprls to retrieve gnat installation path and see
            # if we find the file somewhere in the project search path
            if not self.is_native:
                res = subprocess.check_output(
                    ['gprls', '-v', '--target=%s' % self.tgt.target],
                    stderr=subprocess.STDOUT).decode()
                print("gprls: " + str(res))
            else:
                res = subprocess.check_output(
                    ['gprls', '-v'],
                    stderr=subprocess.STDOUT).decode()
                print("gprls: " + str(res))
            in_prj_search_path = False
            ret = None
            for line in res.splitlines():
                if not in_prj_search_path:
                    if line == 'Project Search Path:':
                        in_prj_search_path = True
                    continue
                line = line.strip()
                if line == '<Current_Directory>':
                    continue
                if len(line) == 0:
                    break
                tentative = os.path.join(line, rts_json_file)
                if os.path.exists(tentative):
                    ret = os.path.normpath(tentative)
                    break
        print(str(ret))
        assert ret is not None, "Cannot find %s" % rts_json_file
        return SharedRTSSources(ret)
```

BrentSeidel, 2020-10-21:
I expected that it was looking in those directories.  The thing is that that installed the arm-elf in its own directory alongside the GNAT community directory.  Thus, I have .../GNAT/2020/ and .../GNAT/2020-arm-elf/.  So, in .../GNAT/2020/ I created a symlink:

ln -s ../2020-arm-elf/arm-eabi arm-eabi

and the build process gets further.  Now, I get the following:
./build_rts.py --output=temp --build sam4s
sam4s
install runtime sources for sam4s
building project /Volumes/External - Data/Users/brent/Development/Ada/bb-runtimes/temp/zfp-sam4s/runtime_build.gpr
Setup
   [mkdir]        object directory for project Runtime_Build
   [mkdir]        library directory for project Runtime_Build
Compile
   [Asm_Cpp]      breakpoint_handler-cortexm.s
   [Ada]          i-c.ads
   [Ada]          s-textio.adb
   [Ada]          s-memory.adb
   [Ada]          board_config.ads
   [Ada]          gnat.ads
board_config.ads:28:06: "Interfaces.Sam" is not a predefined library unit
board_config.ads:29:06: "Interfaces.Sam.Pmc" is not a predefined library unit

   compilation of board_config.ads failed

gprbuild: *** compilation phase failed
Traceback (most recent call last):
  File "./build_rts.py", line 274, in <module>
    main()
  File "./build_rts.py", line 261, in main
    subprocess.check_call(cmd)
  File "/opt/local/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/subprocess.py", line 363, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['gprbuild', '-j0', '-p', '-P', '/Volumes/External - Data/Users/brent/Development/Ada/bb-runtimes/temp/zfp-sam4s/runtime_build.gpr']' returned non-zero exit status 4.

Fabien-Chouteau, 2020-10-22:
It's ok to have the two install side by side, this is what I do.
I think you should start from a clean install and do not add the symlink.

closed 2020-10-22 by Fabien-Chouteau