AdaCore/bb-runtimes #43: Runtime does not compile when Has_FPU is set to False

NicoPy, 2020-11-20:
Hi,

To do some tests, I need to compile the runtime (ravenscar full) with soft Float instead of hard Float.
Here is what I did  :
- set _Has_FPU_ to False in s-bbpara__xxxx.ads. 
- make _has_fpu()_ property return False (cortexm.py)
- make _has_single_precision_fpu()_ property return False in (cortexm.py)
- make _has_double_precision_fpu()_ property return False in (cortexm.py)
- remove the _-mfpu=_ switch in _compiler_switches()_ (cortexm.py)
- add the _-msoft-float_ switch in _compiler_switches()_ (cortexm.py)

Setting _Has_FPU_ to False makes the runtime compilation fail :
```
C:\Users\Nicolas\AppData\Local\Temp\cctSKb2B.s: Assembler messages:
C:\Users\Nicolas\AppData\Local\Temp\cctSKb2B.s:83: Error: selected FPU does not support instruction -- `vstmdbeq r12!,{s16-s31}'
C:\Users\Nicolas\AppData\Local\Temp\cctSKb2B.s:94: Error: selected FPU does not support instruction -- `vldmiane r12!,{s16-s31}'
```
This error is not a surprise since I instructed the compiler to not use hard float instructions (_-msoft-float_ switch). 

However, setting _Has_FPU_ to False makes the runtime compilation fail :
```
s-statxd.adb:396:22: construct not allowed in configurable run-time mode
s-statxd.adb:396:22: file s-fatflt.ads not found
s-statxd.adb:396:22: entity "System.Fat_Flt.Attr_Float" not available
```
When I fix this error by manually disabling Streams, the same error rises again somewhere else.

How can I fix this problem ?

NicoPy, 2020-11-22:
I have understood how FPU configuration works (for ARM architecture).
Here it is for reference.

_cortexm.py_ is used to configure the runtime sources/configuration :
- _has_fpu()_ property must **always** be set to _True_ else the runtime does not compile. 

- To use ***software floating point*** calculations, _has_single_precision_fpu()_ **and** _has_double_precision_fpu()_ properties must return ***False***. You also have to set ***-msoft-float*** switch in _compiler_switches()_.

- To use ***hardware floating point*** calculations, _has_single_precision_fpu()_ or _has_double_precision_fpu()_ or both must return ***True***. You also have to set ***-mfpu=xxx*** switch to the correct value in _compiler_switches()_.

In _s-bbpara__xxxx.ads_ the constant _Has_FPU_ is indirectly related to the runtime configuration. 
When _Has_FPU_ is _True_, the floating point context is saved/restored in _Pend_SV_Handler()_ (_s-bcpcst__pendsv.adb_).
- To use ***software floating point*** calculations, set _Has_FPU_ to ***False***.
- To use ***hardware floating point*** calculations, set _Has_FPU_ to ***True***.

<br/>

To be more complete :
|_has_double_precision_fpu()_ | _has_single_precision_fpu()_ | --> |  | compiler switch | _Has_FPU_ |
| --- | --- | --- | --- | --- | --- |
| False | False | --> | soft float | -msoft-float | False |
| False | True  | --> | single precision : hard <br/>double precision : soft | -mfpu=xxx | True |
| True  | False | --> | single precision : soft <br/>double precision : hard | -mfpu=xxx | True |
| True  | True  | --> | single precision : hard <br/>double precision : hard | -mfpu=xxx | True |

Fabien-Chouteau, 2020-11-23:
Hi @NicoPy,
I am glad you found the answer :)

closed 2020-11-23 by Fabien-Chouteau