AdaCore/bb-runtimes #52: Overhead analysis on Ravenscar for zynq-7000

BottCode, 2021-03-23:
Hello.

In order to make a [fine-grained response time analysis (section 3)](https://www.math.unipd.it/~tullio/RTS/2012/RTSv29y2005.pdf) of real-time applications on Ravenscar kernel, I need to study the overhead timing behaviour of the runtime.
A parameter that I need to compute is the overhead due to the clock handler. Such parameter depends on the system tick frequency, i.e. the period of the interrupt request from the timer. Such period should define the unit of time in the system, i.e. the minimum interval of time handled by the kernel. 

In package GNAT CE 2019 that I've (not so recently) downloaded form AdaCore website, I can see that
```
package System.BB.Parameters is
   pragma Pure;

   --------------------
   -- Hardware clock --
   --------------------

   Clock_Frequency : constant := 333_000_000;
   --  Frequency of the CPU clock in Hz. We hard-code this hear to allow static
   --  computation of the required prescaler.

   Ticks_Per_Second : constant := Clock_Frequency;
```

while [on this repo](https://github.com/AdaCore/bb-runtimes/blob/19-sustained/src/s-bbpara__zynqmp-el2.ads#L50) such values are differents. Why?

Apart from this, should I infer that the unit of time in Ravenscar zynq-7000 is `1/333 = 3 nanoseconds`? It seems extremely short/fast. Does it mean that the clock is updated every `3 ns`? Am I using the corrects parameters?

Fabien-Chouteau, 2021-03-23:
Hello @BottCode,

The Ravenscar run-time on the zymc-7000 is tick-less, there is no periodic interrupts from the timer.
The `Ticks_Per_Second` value is used for the `Ada.Real_Time.Time_Unit` definition: https://www.adaic.org/resources/add_content/standards/12rm/html/RM-D-8.html

closed 2021-03-23 by Fabien-Chouteau

BottCode, 2021-03-24:
Nice, thank you very much!