AdaCore/bb-runtimes #68: Generate a source tree for STM32F407 target

patroclo77, 2024-08-18:
Hello,
I am struggling to get a run-time generation for the STM32F407. 

The original sources from which I want to spin out the run time are located at:

/home/adalab/.alire/arm-eabi/lib/gnat/embedded-stm32f4

and the clone of the repo is at /home/adalab/.alire/bb-runtimes.

I am not very concerned about the output directory. Can you please provide a working command with this parameters?

Thanks.

jklmnn, 2024-08-20:
Hi,

if I understand your question correctly you want to build a *new* runtime that is not in bb-runtimes itself, correct?

patroclo77, 2024-08-20:
Yes, the new run-time should take as input the content of following folder:
[image]

Additionally you can also provide the command to generate a runtime for the stm32f407 if such a runtime can be generated as part of the bb-runtimes.  For the sake of testing both versions.

jklmnn, 2024-08-20:
> Additionally you can also provide the command to generate a runtime for the stm32f407 if such a runtime can be generated as part of the bb-runtimes. For the sake of testing both versions.

From what I can tell, there currently is no `stm32f407` runtime in bb-runtimes. To rebuild a runtime from alire with bb-runtimes I suggest you use this [fork](https://github.com/Fabien-Chouteau/bb-runtimes/tree/gnat-fsf-14) which contains changes specific to the alire toolchain. It also contains a file `rts-sources.json` which you need to rebuild the runtimes. For example to rebuild the `stm32f4` runtime you can use
```
./build_rts.py --rts-src-descriptor gnat_rts_sources/lib/gnat/rts-sources.json --build stm32f4
```
`gnat_rts_sources` is contained in the fork I linked above.

Now for your new runtime I suggest you do a runtime extension. You can add the source directories of a different runtime to your new runtime by adding them to `ada_source_path`. If you put your own source directories first the gprbuild will take your files and only add the files of the other runtime that are not in your sources. I have an example of what such a runtime looks like [here](https://github.com/jklmnn/esp-idf-ada-bindings/tree/esp_idf/rts). In this case `ada_source_path` is generated by a script before the runtime is built, so I add its contents here for clarity:
```
src
/home/user/.local/share/alire/toolchains/gnat_riscv64_elf_11.2.4_f4a370b5/riscv64-elf/lib/gnat/zfp-rv32imac/gnat
```
Gprbuild will use the sources from `src` first and then those of the referenced runtime. If you want to exclude any sources from the original runtime you can take a look at `rts.gpr` in the repo I referenced.

All you need for your new custom runtime are
 - `ada_source_path`
 - a gpr project file
 - your runtime sources
 - `runtime.xml` (you should be able to copy this from the original runtime)
 - `ada_target_properties` (you can also copy this)

If you build the runtime with gprbuild you should end up with a ready to use runtime.

Please let me know if anything is unclear.

patroclo77, 2024-08-20:
Hi,
The first line allowed to get what I was looking for. Thanks a lot for your help, very appreciated.

closed 2024-08-20 by patroclo77