AdaCore/bb-runtimes #69: KeyError while generating RV32I runtime

guillermovidalsule, 2024-09-23:
I am trying to generate a new runtime using the bb-repository with the following command: 

`./build_rts.py --output=temp --force --build rv32i`

However I get an error I cannot manage to solve:

`install runtime sources for rv32i
WARNING: replacing a previously existing runtime
  /home/guillermo/Documents/TFG/RISCV32/BB-Test/bb-runtimes/temp/light-rv32i
Traceback (most recent call last):
  File "/home/guillermo/Documents/TFG/RISCV32/BB-Test/bb-runtimes/./build_rts.py", line 465, in <module>
    main()
  File "/home/guillermo/Documents/TFG/RISCV32/BB-Test/bb-runtimes/./build_rts.py", line 435, in main
    projects += installer.install(
  File "/home/guillermo/Documents/TFG/RISCV32/BB-Test/bb-runtimes/support/bsp_sources/installer.py", line 292, in install
    build_flags["gnat_version"] = runtime_sources.version
  File "/home/guillermo/Documents/TFG/RISCV32/BB-Test/bb-runtimes/support/bsp_sources/installer.py", line 51, in version
    return self.cnt["version"]
KeyError: 'version'
`

I have GNAT 10.5.0 installed and I am using linux Ubuntu 22.04.

I want to make a different Python class to support another board, but I cannot even generate the RV32I (which I did not modify).

jklmnn, 2024-09-24:
The error you're encountering is caused by a mismatch between the version of GNAT and bb-runtimes. I assume you're using the FSF GNAT based on GCC 10. In that case you should be able to use https://github.com/Fabien-Chouteau/bb-runtimes/tree/gnat-fsf-10.

guillermovidalsule, 2024-09-24:
Thanks a lot for the quick reply.

Although this repository you directed me to does not produce the KeyError I am still getting `line 136, in _find_rts_sources
    assert ret is not None, "Cannot find %s" % rts_json_file
 AssertionError: Cannot find rts-sources.json`, which seems to be related to the GNAT version (https://forum.ada-lang.io/t/ada-drivers-library/258). Some platforms like Spike work, but most of them produce the aforementioned error

What GNAT version should I download to avoid these compatibility errors?

jklmnn, 2024-09-25:
You may have to point bb-runtimes to the `rts-sources.json` manually by using the `--rts-src-descriptor` in `build_rts.py`. Where this file is located depends on the distribution of your GNAT. If you're using the GNAT from [alire](https://github.com/alire-project/alire) which I would recommend to use it should work with the repository I mentioned above (assuming you're using the same GNAT version from alire and the repository). Given that alire doesn't provide a GNAT 10.5 I assume you're using the GNAT shipped with Ubuntu?

guillermovidalsule, 2024-09-26:
I used the `rts-sources.json` option and it worked perfectly. Nonetheless, I will also test alire like you mentioned.

Thank you!

PD: yes, I am using the GNAT shipped with Ubuntu.

guillermovidalsule, 2024-11-14:
I will reply to this issue instead of opening one because it is highly related.

I recently decided to use a more recent version of the [repository](https://github.com/Fabien-Chouteau/bb-runtimes/tree/gnat-fsf-10) you referenced. I chose the latest [version](https://github.com/Fabien-Chouteau/bb-runtimes/tree/gnat-fsf-14), gnat-fsf-14 and installed Alire, like you suggested.

Therefore, I downloaded Alire, updated GNAT and updated GCC.
-  `gnat --version` : 14.2.0
- `gcc --version` : 14.2.0
- `alr --version` : 2.0.2+9b80158
- `gprbuild --version` : GPRBUILD Community 2021 (20210519) (x86_64-pc-linux-gnu)

I tried adding a line with the version inside the installer python script, but then I get the same compilation error whenever I want to build any generated runtime. 

Thanks in advance.

jklmnn, 2024-11-14:
Are you sure it is using the correct version? Your `gprbuild` reports to be GNAT Community 2021 so it could be it's still trying to use the wrong toolchain somewhere. In any case I'll take a look at it.

guillermovidalsule, 2024-11-15:
I added the `gprbuild` from Alire to the PATH variable and now it reports `GPRBUILD 22.0.0 (2021-11-09) (x86_64-pc-linux-gnu)`.

I still have GNAT Community installed at /opt/GNAT so I can use still use it if I need to, but my PATH variable prioritizes the aforementioned versions so I can now work with FSF 14.

Should I remove GNAT Community completely? When downloading Alire and the FSF 14 I followed the official [guide](https://alire.ada.dev/transition_from_gnat_community.html) and updated the PATH variable.

guillermovidalsule, 2024-11-15:
I found the problem.

As you mentioned I was selecting the wrong toolchain. Whilst configuring Alire I added to the PATH variable the `gnat_native` bin folder (else I would get other errors when generating the runtimes like `gnatls not found,` even when previously using `alr toolchain --select`). The problem is that when I tried to generate RISC-V targets it kept using that toolchain instead of the riscv one (for example).

I fixed it by modifying the PATH to include the RISC-V version of GNAT and removing `gnat_native`, which is not very effective but works.

Thanks for the help!

closed 2024-11-15 by guillermovidalsule