AdaCore/bb-runtimes #71: 2_ENVIRONMENT.md may need updating

rogermc2, 2024-12-05:
2_ENVIRONMENT.md  specifies the command :
`$ ./build_rts.py --bsps-only --output=build --prefix=arm-eabi/lib/gnat --link stm32f4`
However running this command results in:
```
./build_rts.py --bsps-only --output=build --prefix=arm-eabi/lib/gnat --link stm32f4
usage: build_rts.py [-h] [-v] [-f] [--rts-src-descriptor RTS_SRC_DESCRIPTOR] [--gen-doc] [--compiler COMPILER] [-o OUTPUT] [-l] [-b] [--build-flags BUILD_FLAGS] [--shared]
                    [--profiles PROFILES]
                    target [target ...]
build_rts.py: error: unrecognized arguments: --bsps-only --prefix=arm-eabi/lib/gnat
```

What should be the correct command?

jklmnn, 2024-12-05:
You are right the command is no longer available as the documentation is unfortunately out of date. Unfortunately there is no single command that fixes the problem as some of the assumptions the documentation makes no longer hold. I have written a https://github.com/AdaCore/bb-runtimes/issues/68#issuecomment-2298590451 on another issue on how to build a custom runtime. Let me know if that works for you as a workaround.

rogermc2, 2024-12-12:
Many thanks. 
Following advice from #68 I ran ./build_rts.py --rts-src-descriptor gnat_rts_sources/lib/gnat/rts-sources.json --build stm32f4 successfully:
runtimes successfully installed in /System/Volumes/Data/Ada_Source/bb-runtimes/install.

My only problem was in learning how to "clone" a fork! For which chatGPT provided the necessary instructions.
Now to try to do a runtime extension.